FreeRTOS/FreeRTOS-Cellular-Interface #16: Possible Null Pointer Dereference

ACG-Jake, 2021-07-06:
https://github.com/FreeRTOS/Lab-Project-FreeRTOS-Cellular-Library/blob/4681d3998d6bcbd333d6b605dd83eb21d2c979f3/common/src/cellular_pktio.c#L379

I noticed in the amazon-freertos repo(line 381 in that repo's version of cellular_pktio.c) that there is a null pointer check before dereferencing "pContext->pCurrentCmd"
( ( pContext->pCurrentCmd == NULL ) ? "NULL" : pContext->pCurrentCmd ),

Diving through the history, that repo updated the check in commit 1fdb31c2f94af2bc3cde30cd14901f04db866d10 on November 18th, 2020 1:07am. A little bit after this repo was created. There were additional null pointer dereference checks added in that commit.

All subsequent commits in the other repo seem to be targeted at adding support for more devices.

chinglee-iot, 2021-07-09:
Thank you Jake,
You are right. These null pointer check to print log should be merged back to the repository. We will update it soon. Thank you for your discovery and reminding us.

chinglee-iot, 2021-07-24:
#22 Fix this issue as well as other possible pContext->pCurrentCmd Null pointer.
Thank you for your feedback.

closed 2021-07-24 by chinglee-iot