FreeRTOS/FreeRTOS-Cellular-Interface #18: unknown type name 'BOOL'

matsujirushi, 2021-07-21:
I got below errors:
```
../Lib/cellular/modules/sara_r4/cellular_r4.h:117:5: error: unknown type name 'BOOL'
../Lib/cellular/modules/sara_r4/cellular_r4.h:128:5: error: unknown type name 'BOOL'
../Lib/cellular/modules/sara_r4/cellular_r4.h:129:5: error: unknown type name 'BOOL'
../Lib/cellular/modules/sara_r4/cellular_r4_api.c:1053:72: error: 'TRUE' undeclared (first use in this function)
../Lib/cellular/modules/sara_r4/cellular_r4_api.c:1260:97: error: 'TRUE' undeclared (first use in this function)
../Lib/cellular/modules/sara_r4/cellular_r4_api.c:1419:68: error: 'FALSE' undeclared (first use in this function)
../Lib/cellular/modules/sara_r4/cellular_r4_api.c:2435:94: error: 'TRUE' undeclared (first use in this function)
```

chinglee-iot, 2021-07-23:
Hi Matsujirushi,

Thank you for your feedback. We will read your PR and fix this issue.

chinglee-iot, 2021-07-24:
Your PR is reviewed and merged.
Thank you for your contribution to cellular library.

closed 2021-07-24 by chinglee-iot