FreeRTOS/FreeRTOS-Cellular-Interface #50: 'uint8_t' does not name a type

matsujirushi, 2021-11-04:
Hi,
I got these errors when my code include cellular library headers only.

**Incude:**
```
#include <cellular_config.h>
#include <cellular_config_defaults.h>
#include <cellular_comm_interface.h>
```

**Error:**
```
../Lib/FreeRTOS-Cellular-Interface/source/include/cellular_types.h:416:5: error: 'uint8_t' does not name a type
```

I think it would be better to include stdint.h here.
https://github.com/FreeRTOS/FreeRTOS-Cellular-Interface/blob/main/source/include/cellular_types.h#L39

chinglee-iot, 2021-11-04:
Thank you for the feedback. I will fix this ASAP.

chinglee-iot, 2021-11-05:
The issue is fixed in this PR. Once it is merged I will close this issue.
https://github.com/FreeRTOS/FreeRTOS-Cellular-Interface/pull/53

closed 2021-11-07 by chinglee-iot