FreeRTOS/FreeRTOS-Cellular-Interface #59: strnlen compile error with Arm Compiler 6.16

renesas-adam-benson, 2021-11-11:
When testing with Arm Compiler 6.16 I see the following compile error:

`Error: L6218E: Undefined symbol strnlen (referred from cellular_at_core.o)`

Unless I'm mistaken, I don't believe AC6 supports strnlen. I've temporarily resolved the compile issue by defining strnlen in the `cellular_platform.h` file like so:

```c
#if defined(__ARMCC_VERSION)           /* AC6 compiler */
inline size_t strnlen (const char * s, size_t maxlen)
{
    const char * p;

    p = memchr(s, '\0', maxlen);
    if (p == NULL)
    {
        return maxlen;
    }

    return (size_t) (p - s);
}

#endif
```

chinglee-iot, 2021-11-16:
Hi Adam,

Thank you for your feedback.
We will have internal discussion about the compile error with AC6 compiler.
Once we have conclusion, we will update in this issue.

chinglee-iot, 2021-11-17:
Hi Adam,

Currently, we suggest the same method you use to resolve this compilation error.

> Developers can provide strnlen in cellular_platform.h if strnlen is not supported in their compilation environment.

If you have suggestion about this, please share with us.
We would like to know more from you.

chinglee-iot, 2021-11-22:
Hi Adam,

I will close this PR.
If your have further suggestion, please reopen this issue and comment.

chinglee-iot, 2021-11-24:
Hi Adam,

We are trying to remove the unnecessary usage of strnlen in #64.
Once merged, I will update this issue.

closed 2021-12-03 by chinglee-iot