FreeRTOS/FreeRTOS-Cellular-Interface #66: Many arguments in LogDebug lines

mrcamilletti, 2021-11-26:
Hi! I'm porting the library to a Nordic nrf52 chip. I've some issues with these lines, where the log command has more arguments than permitted by the SDK log function (the maximum number is 6). Is it possible to divide it in two or more calls ?

https://github.com/FreeRTOS/FreeRTOS-Cellular-Interface/blob/c4ccd2c09f6bcbd79ce31e773a35f552f8c3f16a/source/cellular_3gpp_api.c#L488

https://github.com/FreeRTOS/FreeRTOS-Cellular-Interface/blob/c4ccd2c09f6bcbd79ce31e773a35f552f8c3f16a/source/cellular_3gpp_api.c#L1795

AniruddhaKanhere, 2021-11-26:
Hello @mrcamilletti, thanks for taking the time to report this. We'll take a look and get back to you.

chinglee-iot, 2021-11-29:
@mrcamilletti, @AniruddhaKanhere 
Thank you the your feedback. I will follow this issue.

chinglee-iot, 2021-12-01:
@mrcamilletti, @AniruddhaKanhere 
This issue is fixed in #67.

closed 2021-12-01 by chinglee-iot