FreeRTOS/FreeRTOS-Cellular-Interface #74: [Feature Request] Secure Sockets Transport Implementation

adpjhu, 2021-12-19:
It would be useful to offer a secure Secure Sockets Transport Implementation. Quetel, UBlox and Sierra Wireless support TLS sockets, thereby allowing the use of certificates stored within the modem and reducing the MCU resources needed.

chinglee-iot, 2021-12-20:
Hi adpjhu,

Thank you for your interest in FreeRTOS-Cellular-Interface.
To offload TLS to cellular modem, new APIs will be required.
We will discuss this feature. Once we have some update, I will reply to this issue.

es-kaga, 2022-06-29:
@awsblake Would you have any idea when this feature will be ready?

chinglee-iot, 2023-02-16:
Currently, we don't have enough resource to propose and implement this feature in the short time.
We add a help wanted label to this issue. Contribute from community can be great help to this project.

diogosilva0307, 2023-07-31:
Hey! 

At Stratio we're trying to use CellularInterface and offload TLS configuration to the modem (BG96).

Now... A few things we've noticed:
1) There's no way to tell the implementation wether to open a secure socket or a regular one.
2) There's no way to issue AT commands that require sending data to the modem (e.g. File upload AT+QFUPL)

Our proposal would be along the lines of:
1) Alter Cellular_SocketSetSockOpt to add support for a new SSL option whose value is the SSL context ID. This way the implementation would be notified of the need for a new SSL socket.

2) Add a new function to the CellularInterface API called Cellular_UploadFile(const uint8_t *pFilename, const uint8_t *pContent, const uint32_t size). This function would then be implemented by the community ports, probably using _Cellular_AtcmdDataSend.

WDYT? RFC

chinglee-iot, 2023-08-01:
Hi diogosilva0307,

> 1. Alter Cellular_SocketSetSockOpt to add support for a new SSL option whose value is the SSL context ID. This way the implementation would be notified of the need for a new SSL socket.

New socket protocol type and socket opt can be added if it can help user to create their own secure socket API in the port. Please feedback your suggestion or raise a PR.

> 2. Add a new function to the CellularInterface API called Cellular_UploadFile(const uint8_t *pFilename, const uint8_t *pContent, const uint32_t size). This function would then be implemented by the community ports, probably using _Cellular_AtcmdDataSend.

AT+QFUPL is similar with socket send command and it is possible to use common layer API **_Cellular_AtcmdDataSend** to implement the Cellular_UploadFile function.

Thank you for your interest in cellular project. If you have problem implementing APIs, please feedback your question. We will discuss with you here.