FreeRTOS/FreeRTOS-Cellular-Interface #81: Setting UDP Local Port

renesas-adam-benson, 2022-01-11:
Are there plans to allow `localPort` in `CellularSocketContext_t` to be set via an API (maybe via `Cellular_SocketSetSockOpt`)? Currently we are setting it directly through the `CellularSocketHandle_t`.

chinglee-iot, 2022-01-12:
Hi Adam,

TCP client socket is supported in current version. The port is automatically assigned by modem.
Can you give us more information about the use case?
We can have more discussion about the feature.

renesas-adam-benson, 2022-05-12:
@chinglee-iot apologies for the very delayed response. For the RYZ014A modem we are adding support for UDP sockets require the local port to be set. The Cellular library already has `localPort` in `CellularSocketContext_t` which is what we use in the AT command. There are just no APIs available for users to set the value of it and the value doesn't seem to be initialized.

ActoryOu, 2022-06-02:
Hi Adam,
We provide the enum to support setting TCP local port by Cellular_SocketSetSockOpt.
Please get the patch #88 and set the local port by the command below.

- Cellular_SocketSetSockOpt( cellular, socket, CELLULAR_SOCKET_OPTION_LEVEL_TRANSPORT, CELLULAR_SOCKET_OPTION_SET_LOCAL_PORT, (uint16_t*)port, sizeof(uint16_t) );

On BG96 and HL7802, we can set the local port successfully. 
In SARA_R4, it returns CELLULAR_UNSUPPORTED because SARA_R4 doesn't support this.

BTW, when we test it in our environment, there is a NAT in cellular network (CHT in Taiwan). 
It means the source IP/port is different if we want to connect to the host which is outside of this core network.

Thanks for your feedback.

closed 2022-06-02 by ActoryOu