FreeRTOS/FreeRTOS-Cellular-Interface #83: _parseSignalQuality function does not compute RSSI

jpolonsky-tyc, 2022-03-03:
In cellular_hl7802_api.c on line 1080, the function _parseSignalQuality calls _Cellular_ConvertCsqSignalRssi and passes the second argument as "berValue", whereas the function expects raw RSSI as "pRssiValue". 
On line 1066, the first token (which is actually the RSSI value) is deliberately skipped.
In effect, it skips RSSI and miscalculates BER.

jpolonsky-tyc, 2022-03-03:
I have also noticed that the callback uses _Cellular_ConvertCsqSignalRssi which expects a response received from "AT+CSQ", while the command used for Cellular_GetSignalInfo is "AT+CESQ". The format of RSSI is different in those two messages.

chinglee-iot, 2022-03-14:
Hi Jpolonsky,

Sorry for late reply. We will look into the question and reply.

chinglee-iot, 2022-03-21:
@jpolonsky-tyc 
This issue is fixed in this PR #84. Thank you for your feedback.

closed 2022-03-21 by chinglee-iot