FreeRTOS/FreeRTOS-Cellular-Interface #91: [BUG] Sara R4 socket opening race condition

es-kaga, 2022-06-16:
I am using Sara R4 module with the FreeRTOS library I noticed that when I am opening the socket using `Cellular_SocketConnect` then I got URC callback registered by the `Cellular_SocketRegisterSocketOpenCallback` before exiting `Cellular_SocketConnect` which leads to race condition on the socket state `socketHandle->socketState` since it is set to `SOCKETSTATE_CONNECTING` despite the fact that socket is already connected. I am not able to send any data on the socket because send function has a check against socket state.

I checked that it happens when cellular module return URC(+UUSOCO: 0,0) just right after response to the `AT+USOCO=` and the calling thread has lower priorioty then the cellular library receive thread then the URC handler is executed before exiting `Cellular_SocketConnect` and socket state is invalid.

The temporary workaround seems to be increasing the priority of thread calling `Cellular_SocketConnect` and setting it back after this function execution.

Could you take a look into this issue ?

chinglee-iot, 2022-06-17:
@es-kaga 
Thank you for your feedback. We will look into this problem and reply later.

chinglee-iot, 2022-06-22:
@es-kaga 
I reproduced the problem you described by adding delay[ before this line](https://github.com/FreeRTOS/FreeRTOS-Cellular-Interface/blob/main/modules/sara_r4/cellular_r4_api.c#L920).

Socket connected URC should only be received after successfully sending the AT command to modem.
[PR #92](https://github.com/FreeRTOS/FreeRTOS-Cellular-Interface/pull/92) move the line to set CONNECTING state before sending the AT command. The state is reverted if the AT command failed.

Once this PR is merged. I will update the issue again.

chinglee-iot, 2022-06-23:
The PR is merged. Thank you for your feedback.
If there is further question related to this issue, please reopen this issue.

closed 2022-06-23 by chinglee-iot