FreeRTOS/FreeRTOS-Cellular-Interface #107: _searchCompareFunc incompatible to bsearch

eelaix, 2022-09-23:
source/cellular_pkthandler.c: In function '_atParseGetHandler':
source/cellular_pkthandler.c:447:60: warning: passing argument 5 of 'bsearch' from incompatible pointer type [-Wincompatible-pointer-types]
  447 |                                                            _searchCompareFunc );
      |                                                            ^~~~~~~~~~~~~~~~~~
      |                                                            |
      |                                                            int32_t (*)(const void *, const void *) {aka long int (*)(const void *, const void *)}
In file included from source/cellular_pkthandler.c:32:
/opt/armgcc/arm-none-eabi/include/stdlib.h:89:24: note: expected '__compar_fn_t' {aka 'int (*)(const void *, const void *)'} but argument is of type 'int32_t (*)(const void *, const void *)' {aka 'long int (*)(const void *, const void *)'}
   89 |          __compar_fn_t _compar);
      |          ~~~~~~~~~~~~~~^~~~~~

chinglee-iot, 2022-09-26:
Hi eelaix,

Thank you for reporting this issue. We will look into this issue and reply.

chinglee-iot, 2022-09-27:
This issue is included in PR #108. Thank you for your feedback.

closed 2022-09-27 by chinglee-iot