FreeRTOS/FreeRTOS-Cellular-Interface #119: Memory leak in Cellular_Cleanup

renesas-adam-benson, 2022-12-12:
The task that is created in `cellular_pktio` on a `Cellular_Init` call is never deleted on `Cellular_Cleanup`/`_Cellular_PktioShutdown`.

https://github.com/FreeRTOS/FreeRTOS-Cellular-Interface/blob/936d4f73b0e162912e3c41595a455319c495f90a/source/cellular_pktio.c#L1187

chinglee-iot, 2022-12-14:
Hi Adam,

Platform_CreateDetachedThread is supposed to delete the thread info after the thread function returns.
The implementation in demo can be referenced.
https://github.com/FreeRTOS/FreeRTOS/blob/391c79958f635ee5476dcf2774dab59e2b151eff/FreeRTOS-Plus/Demo/FreeRTOS_Cellular_Interface_Windows_Simulator/Common/cellular_platform.c#L64-L73

ahmedsayed2014, 2025-03-16:
Hi Adam,

I'm still encountering the same issue—after calling Cellular_Cleanup, the memory reserved for the thread remains allocated in the heap. Could you please let me know if I'm overlooking something?

Thanks!