FreeRTOS/FreeRTOS-Cellular-Interface #123: Receiving a multi-line payload response from MQTT-related BG95 AT commands

mikalai-kavalchuk, 2022-12-20:
Hello!

I'm trying to implement MQTT-related AT commands for BG95 and have encountered one issue related to receiving multi-line response with _AT+QMTRECV_ command.

I registered URC handler for "_QMTRECV_" and can handle incoming MQTT publications.
But if I send MQTT publication with payload that include _\<LF\>_  symbols inside (formatted JSON with line feed):
[image]
thereafter the first _\<LF\>_ symbol in payload interpreted as the end of line so in the URC handler for "_QMTRECV_" I can only see part of the payload up to the _\<LF\>_ symbol. Next parts are come as _AT_UNDEFINED_ messages.

Looks like an issue while processing responses in [_handleAllReceived](https://github.com/FreeRTOS/FreeRTOS-Cellular-Interface/blob/main/source/cellular_pktio.c#L972) because the responses must include **\<CR\>\<LF\>**\<response\>**\<CR\>\<LF\>**  according to [AT Commands Manual](https://www.quectel.com/wp-content/uploads/2021/03/Quectel_BG95BG77BG600L_Series_AT_Commands_Manual_V2.0-3.pdf)

I can register __Cellular_RegisterUndefinedRespCallback_ to proceed all the rest of "_QMTRECV_" but it looks like workaround. What can you suggest to do in this situation? 
Thank you!

chinglee-iot, 2022-12-21:
Hi mikalai,

Thank you for your interest in cellular interface.
I will take a look at the AT command "AT+QMTRECV" and reply to you later.

chinglee-iot, 2022-12-22:
Hi mikalai,
Cellular interface support reading byte stream from AT command.
We can give the following method a try:
* Use the buffer mode for received message : Message not contained in URC. Receive the message from the buffer.
[image]<br>
* Receive the message from buffer with recvID  : Cellular interface can handle one byte stream in the response.
[image]<br>
* Reference the [Cellular_SocketSend ](https://github.com/FreeRTOS/FreeRTOS-Cellular-Interface-Reference-Quectel-BG96/blob/c33a3628e015f88df24b21160ba9dbdabc781da1/source/cellular_bg96_api.c#L2506)implementation for receiving byte stream from AT command response.

mikalai-kavalchuk, 2022-12-22:
Hi chinglee,
Thank you for the information! I will try this mode and report the results later

tpecar-ltek, 2022-12-27:
I would also like to add here that I've encountered the same issue on SIM800C.

There are however other issues with the current MQTT client on SIM800C before I would consider it viable for production use, and I would encourage others to use the FreeRTOS-Cellular socket interface + coreMQTT (or similar) on the SIM800C platform.

---

For the sake of documenting the issue, on SIM800C the incoming messages are provided via `+SMPUBLISH` URC and there is no configurable option in `AT+SMCONF` for a "notification + read at request" mode of operation (as is for example for modem TCP/IP stack).

chinglee-iot, 2022-12-28:
Hi tpecar,

Thank you for your feedback to cellular interface.
It looks like SIM800C doesn't support buffer access mode for MQTT AT command.
Receiving byte stream in URC is a new feature to cellular interface. We would discuss to add this function.

mikalai-kavalchuk, 2022-12-30:
> Hi mikalai, Cellular interface support reading byte stream from AT command. We can give the following method a try:
> 
> * Use the buffer mode for received message : Message not contained in URC. Receive the message from the buffer.
>   [image]
> * Receive the message from buffer with recvID  : Cellular interface can handle one byte stream in the response.
>   [image]
> * Reference the [Cellular_SocketSend ](https://github.com/FreeRTOS/FreeRTOS-Cellular-Interface-Reference-Quectel-BG96/blob/c33a3628e015f88df24b21160ba9dbdabc781da1/source/cellular_bg96_api.c#L2506)implementation for receiving byte stream from AT command response.

Hi chinglee!  I did everything in the same way as for sockets, as you advised, and now I can receive incoming MQTT publications without any issues. 

Thank you so much for your help!

chinglee-iot, 2023-04-11:
We have a PR to support the feature to receive data from URC #137. This PR is still under development.
One of the similar use case is the socket direct push mode. This [PR](https://github.com/FreeRTOS/FreeRTOS-Cellular-Interface-Reference-Quectel-BG96/pull/10) for BG96 can be referenced.
Suggestion to the PR is welcomed.

chinglee-iot, 2023-05-23:
PR #137 is merged. I will close this issue first. Feel free to feedback further question here.

closed 2023-05-23 by chinglee-iot