FreeRTOS/FreeRTOS-Cellular-Interface #124: Typo in CellularContext data structure

tpecar-ltek, 2022-12-28:
Minor nuisance, but still

https://github.com/FreeRTOS/FreeRTOS-Cellular-Interface/blob/b0e43466642136b9bf9e64b756761220f96cf6d8/source/include/private/cellular_common_internal.h#L153-L154

The annoying thing about it is that correcting it might break modem implementations.

In a C11 context (and not compaining in earlier versions unless `-pedantic` is used) it might be possible to make the transition via anonymous union
```c
struct CellularContext
{
    /* ... */

    /* Module Context. */
    union {
        void * pModueContext __attribute__((deprecated));
        void * pModuleContext;
    };
};
```
However, the above example should be reworked if MSVC support is required.

rawalexe, 2022-12-28:
Thank you for reaching out, we are looking into it and will get back to you soon.

chinglee-iot, 2022-12-29:
Hi tpecar,
Thank you for your feedback. This is something we can improve.
pModueContext is an internal variable to cellular common.
Modem porting should use cellular common API "_Cellular_GetModuleContext" to acquire the module context.
So it can be fixed directly without breaking modem implementation. We will fix this problem soon.

tpecar-ltek, 2022-12-29:
Thank you for the clarification. I would like to point out that this anti-pattern is used within the sim70x0 port
https://github.com/FreeRTOS/FreeRTOS-Cellular-Interface-Community-Supported-Ports/search?q=pModueContext

which is why I initially thought it would be a problem. Will open an issue there.

chinglee-iot, 2022-12-29:
We would also update the sim70x0 port so that it works with both version of cellular interface.