FreeRTOS/FreeRTOS-Cellular-Interface #128: Handling commands that report no status information

tpecar-ltek, 2023-01-11:
SIM800C has a fun little command called `AT+CIFSR`

[image]

This command does not provide any status information when it succeeds

[image]

Which is an issue, since it's _mandatory_ to call as a workaround to make Multi-IP TCP/IP start correctly

[image]
[image]

[image]

And now I'm a bit stuck, because all `CellularATCommandType_t` options seem to expect some form of status information, otherwise the library reports timeout.

Any way to handle this case?

tpecar-ltek, 2023-01-12:
My current workaround is to consider this as a data command, and use the prefix parser to set up the response as an extra success token.

Prefix parser, response parser callbacks
```c
static CellularPktStatus_t _parseLocalAddressPrefix( void * pCallbackContext,
                                                     char * pLine,
                                                     uint32_t lineLength,
                                                     char ** ppDataStart,
                                                     uint32_t * pDataLength )
{
    CellularContext_t * pContext = ( CellularContext_t * ) pCallbackContext;

    CellularPktStatus_t pktStatus = CELLULAR_PKT_STATUS_OK;
    uint32_t lineIdx = 0;
    uint32_t argLen  = 0;

    static
    char LocalLine[ CELLULAR_IP_ADDRESS_MAX_SIZE + 1 ] = {'\0'};
    char * pLocalLine = LocalLine;

    if( ( pLine == NULL ) || ( ppDataStart == NULL ) || ( pDataLength == NULL ) )
    {
        pktStatus = CELLULAR_PKT_STATUS_BAD_PARAM;
    }
    else if ( lineLength > CELLULAR_IP_ADDRESS_MAX_SIZE )
    {
        pktStatus = CELLULAR_PKT_STATUS_BAD_RESPONSE;
    }
    else
    {
        /* Expect IP, newline */
        for ( lineIdx = 0; lineIdx < lineLength; lineIdx++ )
        {
            if (( pLine[ lineIdx ] == '\r' ) || ( pLine[ lineIdx ] == '\n' ))
            {
                /* First newline char represents the end of prefix line */
                if ( argLen == 0 ) {
                    pLocalLine[ lineIdx ] = '\0';
                    argLen = lineIdx;
                }
            }
            /* First non-newline data after argLen is unrelated (next URC, etc.) */
            else if ( argLen != 0 ) {
                break;
            }
            /* Otherwise still prefix, copy to local mutable buffer */
            else {
                pLocalLine[ lineIdx ] = pLine[ lineIdx ];
            }
        }

        /* Assuming at least one char found, we consider it valid */
        if (argLen > 0) {
            LogInfo(( "Local IP: %s", pLocalLine ));

            /* TODO: Possibly store it somewhere */

            /* Hack : Parsed line is set up as extra success token, data set to zero
             *        The library will re-parse the line, matching it with our token */
            *pDataLength = 0;

            static
            const char * ExtraTokenTable[] = { LocalLine };

            pContext->tokenTable.pCellularSrcExtraTokenSuccessTable = ExtraTokenTable;
            pContext->tokenTable.cellularSrcExtraTokenSuccessTableSize = 1;
        }
        /* Otherwise wait for full response */
        else {
            pktStatus = CELLULAR_PKT_STATUS_SIZE_MISMATCH;
        }
    }

    return pktStatus;
}

static CellularPktStatus_t _ParseLocalAddress( CellularContext_t * pContext,
                                               const CellularATCommandResponse_t * pAtResp,
                                               void * pData,
                                               uint16_t dataLen )
{
    CellularPktStatus_t pktStatus = CELLULAR_PKT_STATUS_OK;

    /* Clear extra token table, set up by our prefix handler
     *
     * Has to be done within single _Cellular command invocation
     * in order to keep mutexes locked */

    pContext->tokenTable.pCellularSrcExtraTokenSuccessTable = NULL;
    pContext->tokenTable.cellularSrcExtraTokenSuccessTableSize = 0;

    return pktStatus;
}
```

Actual invocation
```c
        CellularAtReq_t atReqActPdn =
        {
            .pAtCmd       = "AT+CIFSR",
            .atCmdType    = CELLULAR_AT_MULTI_DATA_WO_PREFIX,
            .respCallback = _ParseLocalAddress
        };

        cellularStatus = _Cellular_TranslatePktStatus(
                _Cellular_TimeoutAtcmdDataRecvRequestWithCallback(
                        pContext, atReqActPdn, PACKET_REQ_TIMEOUT_MS, _parseLocalAddressPrefix, (void *) pContext ));
```

If there's a cleaner approach, I'm open to suggestions.

tpecar-ltek, 2023-01-17:
Same issue is with the SIM800C's TCP/IP stack commands.

The `AT+CIPSEND` data command responds with `<socket ID>, SEND OK` on success. The `_Cellular_AtcmdDataSend() -> _Cellular_DataSendWithTimeoutDelayRaw()` sets up the `pContext->PktioAtCmdType` to `CELLULAR_AT_NO_RESULT`, so the only way to match the response is by injecting the custom success token with the abovementioned workaround.

Again, if there's a better approach for this issue, I'm open to suggestions.

Alternatively we could discuss on how we could improve the interface to better support this use-case.

tpecar-ltek, 2023-01-17:
Actually, it seems that I can't apply my workaround for _Cellular_AtcmdDataSend(), because that function uses two calls to `_Cellular_AtcmdRequestTimeoutWithCallbackRaw()`
- first one provides the command, which switches the modem into data entry, and we use the prefix handler to match the prompt
- second one provides the actual data, but at that point the prefix handler is NULLed, and command set to CELLULAR_AT_NO_RESULT

chinglee-iot, 2023-02-07:
Hi tpecar,

The command doesn't reply with status information ( OK, ERROR, ... ) is not in the design consideration of cellular interface.
We may need to provide new API to handle this kind of command.
We will discuss about this feature. I'll update this thread once we have some ideas.

chinglee-iot, 2023-05-23:
@tpecar-ltek 

We added two new AT command type to support this kind of AT command in #138.
* **CELLULAR_AT_WO_PREFIX_NO_RESULT_CODE** : if one response is received, this command is regarded as success.
* **CELLULAR_AT_WITH_PREFIX_NO_RESULT_CODE** : if one response with prefix is received, this command is regarded as success.

This PR is still under development. I don't have the modem SIM800C at hand.
It would be great if you are able to give it a try and feedback.

tpecar-ltek, 2023-05-23:
Thank you for working on this issue. I'll check & provide feedback within this week.

tpecar-ltek, 2023-05-29:
I'm currently swamped by other work. Provided that it checks out on your side, you can proceed, and I'll reopen the issue in case of any hiccups.

chinglee-iot, 2023-05-29:
We have unit test cases to verify the implementation. This issue will be closed when the PR is merged.
Thank you again for your feedback.

chinglee-iot, 2023-06-13:
The PR #138 for this issue is merged. Close this issue first.

closed 2023-06-13 by chinglee-iot