FreeRTOS/FreeRTOS-Cellular-Interface #130: _Cellular_HandlePacket() depends on undefined behaviour

tpecar-ltek, 2023-01-17:
`_Cellular_HandlePacket()` > `_processUrcPacket()` -> `urcParseToken()` uses the `strtok_r()` function, and uses its `saveptr`, which isn't intended to be used by the caller.

https://github.com/FreeRTOS/FreeRTOS-Cellular-Interface/blob/936d4f73b0e162912e3c41595a455319c495f90a/source/cellular_pkthandler.c#L132-L175

The function even seems to use the same variable for both `str` and `saveptr`.
The intended end result seems to be a null-terminated `pTokenPtr` containing the URC keyword, and `pSavePtr` containing the arguments to be consumed by the URC handler.

It's probably best that such string manipulation be implemented directly.

chinglee-iot, 2023-02-07:
Hi tpecar,

The implementation try not to use extra memory.
The pInputLine is manipulated to extract the token for URC handler function.
We will discuss another approach to enhancement this function.

chinglee-iot, 2023-02-15:
This issue is updated in PR #131 to manipulate allocated string in function scope in _processUrcPacket only.
Thank you for your feedback. Feel free to reopen this issue if any question.

closed 2023-02-15 by chinglee-iot