FreeRTOS/FreeRTOS-Cellular-Interface #141: Cannot get both IP address of IPv4 and IPv6 when calls Cellular_GetIPAddress  function

KeitaKashima, 2023-07-20:
I use a connectiivitiy module with this cellular lib.
I met a trouble about getting IP address when using SIM card that supports both IPv4 and IPv6.

Cellular_GetIPAddress function only gets IPv4 address when using both IP protocol.
The module returns below IP data. The first one is IPv4, and second one is IPv6.
 "111.112.113.114","1111.2222.3333.4444.5555.6666.7777.8888"

However, Cellular_GetIPAddress function only get "111.112.113.114" of IPv4.

KeitaKashima, 2023-07-31:
@chinglee-iot  Is there any update about this ticket?

chinglee-iot, 2023-07-31:
Sorry for late reply. We look into this issue and fix it. If you already have a fix, you are welcomed to create an PR. We will also discuss with you in the PR.

chinglee-iot, 2023-07-31:
@KeitaKashima 
We create #142 to fix this issue. You are welcome to feedback in the PR or in this thread.
This issue will be closed once it is merged.

KeitaKashima, 2023-08-01:
@chinglee-iot 
Thank you. I am afraid I don't have fixed code.
I will check the #142 ticket.

chinglee-iot, 2023-08-01:
The PR #142 is merged. Thank you for your feedback.

closed 2023-08-01 by chinglee-iot