FreeRTOS/FreeRTOS-Cellular-Interface #149: Cellular_CommonGetRegisteredNetwork API return undefined value

KeitaKashima, 2023-10-02:
Hi, I called the `Cellular_CommonGetRegisteredNetwork` API to test MCC when disconnecting the PDN of the network of cellular.

Then I found the API got the undefined value of MCC.

The reason is that the below pOperatorInfo value is not zero clear after getting Malloc.

`cellularOperatorInfo_t * pOperatorInfo = ( cellularOperatorInfo_t * ) Platform_Malloc( sizeof( cellularOperatorInfo_t ) );`

https://github.com/FreeRTOS/FreeRTOS-Cellular-Interface/blob/7c80e21277e3d214901dbfb68529c21a339e46b7/source/cellular_3gpp_api.c#L1779

The `*pOperatorInfo`  value is copied undefined value below the lines.

https://github.com/FreeRTOS/FreeRTOS-Cellular-Interface/blob/7c80e21277e3d214901dbfb68529c21a339e46b7/source/cellular_3gpp_api.c#L1807

chinglee-iot, 2023-10-02:
@KeitaKashima 
Thank your for reporting this problem. We will look into the problem and discuss with you in this thread.

chinglee-iot, 2023-12-26:
@KeitaKashima 
`Cellular_CommonGetRegisteredNetwork` requires the port to set to the numeric format in `Cellular_ModuleEnableUrc`. If the format is set to other format, for example long or short format, the MCC and MNC won't be updated and the value is undefined. This is probably the problem you encountered.

We address this issue in #161 to set the format in `atcmdUpdateMccMnc` in order not to rely on the port initialization code.

chinglee-iot, 2024-01-02:
PR #161 is merged. Thank you for reporting this issue.

closed 2024-01-02 by chinglee-iot