FreeRTOS/FreeRTOS-Cellular-Interface #150: Wrong formatting / argument type in log traces generate warnings

guillaumeVolery, 2023-10-04:
# Problem description

By building the library with arm-none-eabi-gcc compiler and setting the flag -Werror, the library is producing the following warnings (errors) due to the use of wrong format specifier in the log traces... 

The correction is very simple, use %lu instead of %u, use %ld instead of %d. 

# Awaited result

New release of the library implementing the correction above and no more generating any warning.

# List of reported errors:

- [ ] cellular_common.c:602:20 error: format '%u' expects argument of type 'unsigned int', but argument 2 has type 'uint32_t' {aka 'long unsigned int'}
- [ ] cellular_common.c:655:25: error: format '%u' expects argument of type 'unsigned int', but argument 2 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
- [ ] cellular_common_api.c:110:25: error: format '%d' expects argument of type 'int', but argument 3 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
- [ ] cellular_common_api.c:126:25: error: format '%d' expects argument of type 'int', but argument 3 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
- [ ] cellular_common.c:919:25: error: format '%u' expects argument of type 'unsigned int', but argument 2 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
- [ ] cellular_3gpp_api.c:492:21: error: format '%d' expects argument of type 'int', but argument 2 has type 'int32_t' {aka 'long int'} [-Werror=format=]
- [ ] cellular_pktio.c:145:17: error: format '%d' expects argument of type 'int', but argument 3 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
- [ ] cellular_pkthandler.c:823:29: error: format '%u' expects argument of type 'unsigned int', but argument 2 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
- [ ] cellular_pktio.c:177:17: error: format '%d' expects argument of type 'int', but argument 3 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
- [ ] cellular_pkthandler.c:823:29: error: format '%d' expects argument of type 'int', but argument 3 has type 'int32_t' {aka 'long int'} [-Werror=format=]
- [ ] cellular_pktio.c:186:17: error: format '%lu' expects argument of type 'long unsigned int', but argument 3 has type 'size_t' {aka 'unsigned int'} [-Werror=format=]
- [ ] cellular_pkthandler.c:839:25: error: format '%u' expects argument of type 'unsigned int', but argument 2 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
- [ ] cellular_pktio.c:549:17: error: format '%d' expects argument of type 'int', but argument 4 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
- [ ] cellular_pktio.c:620:25: error: format '%d' expects argument of type 'int', but argument 2 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
- [ ] cellular_pktio.c:673:21: error: format '%d' expects argument of type 'int', but argument 4 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
- [ ] cellular_pktio.c:673:21: error: format '%d' expects argument of type 'int', but argument 5 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
- [ ] cellular_pktio.c:673:21: error: format '%d' expects argument of type 'int', but argument 6 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
- [ ] cellular_pktio.c:888:25: error: format '%u' expects argument of type 'unsigned int', but argument 2 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
- [ ] cellular_pktio.c:888:25: error: format '%u' expects argument of type 'unsigned int', but argument 3 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
- [ ] cellular_pktio.c:1029:21: error: format '%u' expects argument of type 'unsigned int', but argument 2 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
- [ ] cellular_pktio.c:1430:17: error: format '%d' expects argument of type 'int', but argument 2 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]

guillaumeVolery, 2023-10-05:
Hi @moninom1 

Thank you for your reply.

This is actually my problem:
```
git push --set-upstream origin Issue150
remote: Permission to FreeRTOS/FreeRTOS-Cellular-Interface.git denied to guillaumeVolery.
fatal: unable to access 'https://github.com/FreeRTOS/FreeRTOS-Cellular-Interface.git/': The requested URL returned error: 403
```

moninom1, 2023-10-05:
Hi @guillaumeVolery 

You need to create a fork of the repo and then push the changes to your branch in your forked repo. After that you will be able to create a PR to main branch. Please check the following steps :

1: Fork the FreeRTOS-Cellular-Interface repo
2: Clone the forked FreeRTOS-Cellular-Interface repo.
3: Add changes locally and commit it.
4: Push the changes to your own forked GitHub repo.
5: Create a pull request from forked FreeRTOS-Cellular-Interface repo to main repo.

Let me know if this works.

guillaumeVolery, 2023-10-05:
Hi @moninom1 

Thank you for your insights. I have complete the process and you will find the pull request [here](https://github.com/FreeRTOS/FreeRTOS-Cellular-Interface/pull/151).

👍

chinglee-iot, 2023-10-14:
PR #154 is merged. We will close this issue. Feel free to reopen this issue if any further problem.

closed 2023-10-14 by chinglee-iot