FreeRTOS/FreeRTOS-Cellular-Interface #156: It seems that System would be locked or doesn't work properly when Mutex cannot get due to PlatformMutex_Lock doesn't have return value

KeitaKashima, 2023-11-01:
Hi, I wonder the system would be locked or not work properly if PlatformMutex_Lock API cannot get Mutex or happened the timeout of Mutex.

https://github.com/FreeRTOS/FreeRTOS-Cellular-Interface/blob/ccbd0fcce06fcb7b842bed61f207523c8f271fcb/source/cellular_common_api.c#L211

        PlatformMutex_Lock( &pContext->PktRespMutex );    // It gets the Mutex however, the API doesn't have return value. So, the code cannot check timeout.
        pContext->cbEvents.networkRegistrationCallback = networkRegistrationCallback;
        pContext->cbEvents.pNetworkRegistrationCallbackContext = pCallbackContext;
        PlatformMutex_Unlock( &pContext->PktRespMutex );

Does it nerver happnened?

shubnil, 2023-11-02:
HI @KeitaKashima ,
Thanks for bringing this up. We have not observed this.
Do you have a use case where you are seeing this behavior

KeitaKashima, 2023-11-05:
Hi @shubnil ,

Thank you for your reply.
I have never seen the lock of the system here. I just checked the implementation of the Semaphore portion.

closed 2024-05-16 by n9wxu