FreeRTOS/FreeRTOS-Cellular-Interface #164: Incompatible pointer types between CellularAtParseTokenHandler_t and functions in cellular_common_api.h

thirtytwobits, 2024-02-29:
Most of the implementations I've seen of this interface are like [FreeRTOS-Cellular-Interface-Reference-Quectel-BG96/source/cellular_bg96_urc_handler.c](https://github.com/FreeRTOS/FreeRTOS-Cellular-Interface-Reference-Quectel-BG96/blob/f61fc4c9b23668a72868c02f7e77829037e4d7f2/source/cellular_bg96_urc_handler.c#L83) where an array of `CellularAtParseTokenMap_t` is defined using functions like `Cellular_CommonUrcProcessCereg` from [`cellular_common_api.h`](https://github.com/FreeRTOS/FreeRTOS-Cellular-Interface/blob/main/source/include/common/cellular_common_api.h#L277): 

```
CellularAtParseTokenMap_t CellularUrcHandlerTable[] =
{
    { "CEREG",          Cellular_CommonUrcProcessCereg },
    { "CGREG",          Cellular_CommonUrcProcessCgreg },
    { "CREG",           Cellular_CommonUrcProcessCreg  },
    { "POWERED DOWN",   _Cellular_ProcessPowerDown     },
    { "PSM POWER DOWN", _Cellular_ProcessPsmPowerDown  },
    { "QIND",           _Cellular_ProcessIndication    },
    { "QIOPEN",         _Cellular_ProcessSocketOpen    },
    { "QIURC",          _Cellular_ProcessSocketurc     },
    { "QSIMSTAT",       _Cellular_ProcessSimstat       },
    { "RDY",            _Cellular_ProcessModemRdy      }
};
```

If you look at [`CellularAtParseTokenHandler_t`](https://github.com/FreeRTOS/FreeRTOS-Cellular-Interface/blob/main/source/include/common/cellular_common.h#L93), however, you'll see it's defined as a function pointer to a function that does not return a value whereas `Cellular_CommonUrcProcessCgreg` returns a [`CellularError_t`](https://github.com/FreeRTOS/FreeRTOS-Cellular-Interface/blob/main/source/include/cellular_types.h#L87). Are the implementations wrong to use this function in this manner or is there a flaw in the interface design?

chinglee-iot, 2024-03-06:
@thirtytwobits 
URC parsing error is handled in port URC handler implementation. Therefore, it is defined as a function pointer with no return value. `Cellular_CommonUrcProcessCgreg` is common handler implementation for 3GPP URC. The return value helps to indicate the URC parsing status to the port. Cellular port can make use of this common handler with the following example code:
```c
void _Cellular_ProcessCgreg( CellularContext_t * pContext,
                                               char * pInputLine )
{
    CellularPktStatus_t pktStatus;
    pktStatus = Cellular_CommonUrcProcessCgreg( pContext, pInputLine );
    if( pktStatus != CELLULAR_PKT_STATUS_OK )
    {
        /* Handle different packet status error code here. */
    }
}
```
Thank you for creating this issue. We would like to create a PR to update BG96 port with the example above.

chinglee-iot, 2024-04-10:
The BG96 implementation is updated in this PR https://github.com/FreeRTOS/FreeRTOS-Cellular-Interface-Reference-Quectel-BG96/commit/1c71555e690e24c4be3470c2734eee9a24ae07f7. Thank you for reporting this issue.

closed 2024-04-10 by chinglee-iot