FreeRTOS/FreeRTOS-Cellular-Interface #167: CellularPdnConfig_t has const data members.

thirtytwobits, 2024-05-16:
In `cellular_types.h` the `CellularPdnConfig` structure has three `const char` fields making it impossible to use this structure without violating constness. 

```
typedef struct CellularPdnConfig
{
    CellularPdnContextType_t pdnContextType;                   /**< PDN Context type. */
    CellularPdnAuthType_t pdnAuthType;                         /**< PDN Authentication type. */
    const char apnName[ CELLULAR_APN_MAX_SIZE + 1 ];           /**< APN name. */
    const char username[ CELLULAR_PDN_USERNAME_MAX_SIZE + 1 ]; /**< Username. */
    const char password[ CELLULAR_PDN_PASSWORD_MAX_SIZE + 1 ]; /**< Password. */
} CellularPdnConfig_t;
```

`apnName`, `username`, and `password` all need to be changed to `char`.

chinglee-iot, 2024-05-17:
@thirtytwobits 
Thank you for reporting this issue. We create a PR #168 to address this issue. Once it is merged, we will update in this thread again.

chinglee-iot, 2024-05-17:
The PR #168 is merged. Thank you for reporting this issue.

closed 2024-05-17 by chinglee-iot

thirtytwobits, 2024-05-28:
Thanks so much!