FreeRTOS/FreeRTOS-Cellular-Interface #169: Configuration of PKTIO_READ_BUFFER_SIZE

KeitaKashima, 2024-05-21:
Hi, I am using this cellular interface lib, but I faced an issue of below.

The log message showed `No empty space from comm if to handle incoming data, reset all parameter for next incoming data.`

[image]

After I changed `PKTIO_READ_BUFFER_SIZE` of definition value to 4096, it was solved.

https://github.com/FreeRTOS/FreeRTOS-Cellular-Interface/blob/aff376d03a4483d7ecb41df6da8401a6ed7e6a38/source/include/private/cellular_common_internal.h#L51

* Question
  *  Is the size of `PKTIO_READ_BUFFER_SIZE` TCP window size?

  It has comment of  `This should be larger than TCP packet size.`, but I think it is window size of TCP. Is my understanding correct?

* Request
  * Is it possible to add (#ifndef PKTIO_READ_BUFFER_SIZE)  to be possible the configuration by the config.h file?

chinglee-iot, 2024-05-23:
@KeitaKashima
Thank you for reporting this issue.

You're correct in your understanding. The comment should describe the maximum AT command response size, which is typically the TCP socket receive command response packet size. This should be configurable according to the modem specification. We will create a PR to expose this configuration.

chinglee-iot, 2024-05-25:
@KeitaKashima 
We've created a PR to address this issue #173. It would be great if you can also help to test this PR.

KeitaKashima, 2024-05-28:
@chinglee-iot 

Thank you for fixing this portion.

I could redefine the `PKTIO_READ_BUFFER_SIZE` in my config.h.
**And it is working fine!**

＜cellular_config.h＞

[image]

The `pketioReadBuf` value is set correctly as I set `4096+1` .

[image]

chinglee-iot, 2024-05-28:
Thank you for your verification. This issue will be closed. Feel free to reopen this issue or create another one if you have further question.

closed 2024-05-30 by chinglee-iot