FreeRTOS/FreeRTOS-Cellular-Interface #184: Missing Cellular Handles typedefs in Doxygen of Cellular Interface lib v1.4.0

KeitaKashima, 2024-12-05:
Hi There,

I found the Doxygen to lack a description. It's a small issue, but I'm posting this here as other parts may be missing.

Typedef description of `CellularContext_t ` and `CellularHandle_t` are missing from v1.4.0 of doxygen.

* V1.4.0 : NG `CellularContext_t ` and `CellularHandle_t` are missing
[image]
https://freertos.github.io/FreeRTOS-Cellular-Interface/v1.4.0/group__cellular__datatypes__handles.html#ga2a6b2f5ebf6684559ce5865efbdd4770

* V1.3.0 : OK There are three typedef descriptions
[image]
https://freertos.github.io/FreeRTOS-Cellular-Interface/v1.3.0/group__cellular__datatypes__handles.html

These missing typedefs are used int the v1.4.0 code.
[image]

https://github.com/FreeRTOS/FreeRTOS-Cellular-Interface/blob/v1.4.0/source/include/common/cellular_common.h#L260

ActoryOu, 2024-12-10:
Hi @KeitaKashima,
Thanks for reporting this to us! #185 is raised to fix this issue.
This issue is supposed to be fixed when merged.

ActoryOu, 2024-12-18:
Closing since we've released [v1.4.1](https://github.com/FreeRTOS/FreeRTOS-Cellular-Interface/releases/tag/v1.4.1) to update this doc.

closed 2024-12-18 by ActoryOu